medusalix/xow #187: Weak rumble not working (master) for Xbox One controller (1708).

maudemor, 2021-10-03:
So, I just got my controller today, and wanted to test it on Pop OS!

I cloned and decided to run it from master instead of a release, setup the service and paired. I played a couple of games and realized that no vibration was happening... 

When I try using `fftest`, I receive the following output:

```
Uploading effect #0 (Periodic sinusoidal) ... Error:: Invalid argument
Uploading effect #1 (Constant) ... Error: Invalid argument
Uploading effect #2 (Spring) ... Error: Invalid argument
Uploading effect #3 (Damper) ... Error: Invalid argument
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 0)
Uploading effect #5 (Weak rumble, with light motor) ... Error: No space left on device
Enter effect number, -1 to exit
```

And indeed, if I try sending a weak rumble effect, nothing happens.

I ran the DEBUG build, and if I do the following with `fftest`:

```
Enter effect number, -1 to exit
5
Now Playing: Weak Rumble
Enter effect number, -1 to exit
4
Now Playing: Strong Rumble
Enter effect number, -1 to exit
```

I get this as output:

```
$ sudo journalctl -f -u xow
-- Journal begins at Fri 2021-08-27 19:36:28 CDT. --
xow[15404]: 2021-10-03 00:51:55 DEBUG - Channel 157, power: 43
xow[15404]: 2021-10-03 00:51:55 DEBUG - Channel 161, power: 43
xow[15404]: 2021-10-03 00:51:55 DEBUG - Channel 165, power: 43
xow[15404]: 2021-10-03 00:51:55 INFO  - Dongle initialized
xow[15404]: 2021-10-03 00:52:10 INFO  - Controller '1' connected
xow[15404]: 2021-10-03 00:52:11 INFO  - Device announced, product id: 02ea
xow[15404]: 2021-10-03 00:52:11 DEBUG - Firmware version: 3.1.1221.0
xow[15404]: 2021-10-03 00:52:11 DEBUG - Hardware version: 1284.1.1.1
xow[15404]: 2021-10-03 00:52:11 INFO  - Serial number: 02600138633104
xow[15404]: 2021-10-03 00:52:14 INFO  - Battery level: full
xow[15404]: 2021-10-03 00:53:55 DEBUG - Rumble count: 1, duration: 5000, delay: 1000
xow[15404]: 2021-10-03 00:53:55 DEBUG - Rumble strong: 32768, weak: 0, direction: 0
```

So, the rumble event seems to not reach the `void Controller::inputFeedbackReceived` function if it is weak. I'm not knowledgeable on this topic, so I thought to ask, instead of trying to fix it :)

I will test releases later.

kakra, 2021-10-03:
xow doesn't work with ff-memless so it can probably not be tested with fftest because fftest tries to *upload* rumble effects to a memless device - which usually is emulated by ff-memless.

Instead, it skips the unsupported effects (invalid argument), then uploads exactly one effect (strong rumble), and fails the weak rumble because there is no emulated memory storage for the effect. With games directly sending simple rumble commands (instead of real effects), it should just work, tho.

maudemor, 2021-10-03:
I tried with a few different games. I'm not getting weak rumbles to show on the DEBUG output (and of course, the controller is not rumbling). So, for example, when I get hit, I expect a rumble, but I get nothing, and nothing shows in the output.

`fftest` strong rumble works fine. Only the weak rumble does not work... Again, I'm not knowledgeable on this topic, and I only quickly took a look at the code, so not enough to have an idea of what is going wrong.

kakra, 2021-10-03:
If anything works with uploadable effects, only the first upload would work. That's a limitation of xow, it cannot make use of ff-memless. In xpadneo, I'm working on a similar solution for the dongle as xow but with a different approach. It should be able to use ff-memless then.

The games you're talking about: Are those Proton/wine games or native Linux games?

maudemor, 2021-10-03:
I only tried Proton games. I will try a couple of native games I have tomorrow... although I play mostly proton/wine games.