medusalix/xow #194: Controller immediately disconnects after connecting

bfenty, 2021-11-15:
I've successfully paired the controller to the dongle, as soon as the controller pairs, it throws an error and disconnets (and kills the service). Here's the output:

BEFORE connecting
`udo systemctl status xow
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-11-15 17:39:27 CST; 6s ago
   Main PID: 31986 (xow)
      Tasks: 4 (limit: 18988)
     Memory: 796.0K
     CGroup: /system.slice/xow.service
             └─31986 /usr/local/bin/xow

Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 36, power: 43
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 40, power: 43
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 44, power: 42
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 48, power: 42
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 149, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 153, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 157, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 161, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 165, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 INFO  - Dongle initialized
`

AFTER connecting
`sudo systemctl status xow
● xow.service - Xbox One Wireless Dongle Driver
     Loaded: loaded (/etc/systemd/system/xow.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Mon 2021-11-15 17:39:38 CST; 2s ago
    Process: 31986 ExecStart=/usr/local/bin/xow (code=killed, signal=ABRT)
   Main PID: 31986 (code=killed, signal=ABRT)

Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 149, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 153, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 157, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 161, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 DEBUG - Channel 165, power: 40
Nov 15 17:39:27 pop-os xow[31986]: 2021-11-15 17:39:27 INFO  - Dongle initialized
Nov 15 17:39:38 pop-os xow[31986]: terminate called after throwing an instance of 'InputException'
Nov 15 17:39:38 pop-os xow[31986]:   what():  Error opening device: Permission denied
Nov 15 17:39:38 pop-os systemd[1]: xow.service: Main process exited, code=killed, status=6/ABRT
Nov 15 17:39:38 pop-os systemd[1]: xow.service: Failed with result 'signal'.`

here's the lsusb result for the dongle:

`Bus 001 Device 020: ID 045e:02e6 Microsoft Corp. Wireless XBox Controller Dongle`

tjeulink, 2022-03-20:
running xow with elevated privileges fixxed this for me, just fyi in case someone else runs into this.

hcen1997, 2023-01-05:
jast reboot work for me