medusalix/xow #195: make: *** No targets specified and no makefile found. Stop.

whiskeyphilosophy, 2021-11-19:
This is what I get after running make BUILD=RELEASE

The git clone https://github.com/medusalix/xow appears to execute correctly.

kakra, 2021-11-20:
Did you run `cd xow` after cloning?

whiskeyphilosophy, 2021-11-20:
I did not.  Is that something I should know to do after a cloning process
intuitively?  I am new to this.

That did work.  I sincerely appreciate your reply.

kakra, 2021-11-20:
Intuitively? Certainly not. For most somewhat tech savvy users, that is obvious. But it was mentioned before that the documentation lacks to explicitly mention it. It should probably be fixed. OTOH, as a beginner you should probably prefer to use the package provided by your distribution, or ask them to provide one.

We should leave this open so the documentation can be improved.

medusalix, 2022-03-20:
This has been fixed with 6b109c8290ebe61c8ccc3d1960e0d6a7f34aa7c4.

closed 2022-03-20 by medusalix