medusalix/xow #208: Unable to Migrate from `xow` to `xone`

manuth, 2023-04-14:
At time of writing, I find myself unable to migrate from `xow` to `xone`.

I'm using a device which has a built-in Xbox Receiver, which means that there is no hardware button I could press in order to start pairing mode.

To my understanding, unlike `xow`, `xone` doesn't have a feature for programmatically starting pairing mode.

Am I missing something? Does `xone` have such a functionality? If not: Is it planned to add such a functionality to `xone` as well?

Thank you so much for your great work! :smile:

kapsh, 2023-07-26:
xone has low-level pairing control via sysfs. Can you check if commands from https://github.com/dlundqvist/xone#pairing-mode work for your device?

manuth, 2023-07-26:
Awesome, thank you so much, this version of `xone` seems to do the trick!